adafruit/Bluefruit_LE_Connect_Android_V2 #8: Instructions for those new to Android development?

mariedilworth, 2019-07-19:
Could some basic instructions be provided about how to run this code in Android studio? I'm new to Android development and so it's not entirely obvious to me how to integrate this code into an app? Thanks.

mariedilworth, 2019-07-22:
After locating activity_main.xml I am now getting "URI not registered" errors...

<?xml version="1.0" encoding="utf-8"?>
<RelativeLayout xmlns:android="http://schemas.android.com/apk/res/android"
    xmlns:tools="http://schemas.android.com/tools"
    android:id="@+id/container"
    android:layout_width="match_parent"
    android:layout_height="match_parent"
    android:background="@color/background"
    android:orientation="vertical"
    tools:context=".app.MainActivity">

    <FrameLayout
        android:id="@+id/contentLayout"
        android:layout_width="match_parent"
        android:layout_height="match_parent" />

</RelativeLayout>

mariedilworth, 2019-07-22:
Managed to run the app by syncing with Gradle files and updating something in my android studio. Running ok now.

closed 2019-07-22 by mariedilworth

datanonymous, 2019-07-22:
Adjust the source in tools:context