adafruit/Bluefruit_LE_Connect_Android_V2 #22: Tabs dont display on android 11?

bm16ton, 2022-12-03:
Hello, great app i love it! Works great on my android 7 phone, but on android 11 none of the tabs IE "uart" show up. Unfortunatly i dont have access to any other newer android devices so im not sure if its justmy device type thing orif android made breaking changes along the way so i tthought i ask and see if anyone else has similiar issue. Thank you

antonio-openroad, 2022-12-06:
Hi,
Could you provide more info about the peripheral that you are connecting to, and the Android Phone model?

I just tested the app and the UART sections are displayed when connecting to a UART enabled peripheral (Adafruit Bluefruit LE Friend)

Tested on the following devices:
- Samsung Galaxy A40, Android 11
- Pixel 3, Android 12
- GalaxyTab A7 Lite, Android 12
- Pixel 6Pro, Android 13

Screenshot with the sections displayed:
[image]

bm16ton, 2022-12-06:
Thank you so much for responding! I know you must be busy. Its a oneplus 7 pro, the device is seeed xiao nrf52840. I tried all the uart examples from adafruit,nrf etc. But i now know its my phone and not the app, thats the info i needed to move forward. Thank you again kind sir

closed 2023-03-14 by antonio-openroad