adafruit/Bluefruit_LE_Connect_Android_V2 #23: Messages are not being displayed by APP the first time the UART Module is selected

RetiredWizard, 2023-03-02:
Messages sent using the uart_service.write method are not being displayed by the Bluefruit Connect App the first time the UART module is entered. 

As a simple work around if you select the UART module, press the back arrow to return to the module menu and then enter the UART module a second time, messages are displayed as expected.

The following code can be used to demonstrate the issue:
```py
from adafruit_ble import BLERadio
from adafruit_ble.advertising.standard import ProvideServicesAdvertisement
from adafruit_ble.services.nordic import UARTService

from adafruit_bluefruit_connect.packet import Packet

ble = BLERadio()
uart_service = UARTService()
advertisement = ProvideServicesAdvertisement(uart_service)

cmd = ""
while cmd.upper() != "Q":

    if not ble.connected:
        ble.start_advertising(advertisement)
        while not ble.connected:
            pass

    while uart_service.in_waiting and ble.connected and cmd.upper() != "Q":
        ucmd = uart_service.read(uart_service.in_waiting)
        cmd = ucmd.decode().replace("\n","").replace(" ","")
        uart_service.write(("Echoing: ->"+cmd+"<- Enter Q to quit").encode('utf-8'))
        uart_service.write(b'\n')
        print(cmd)
```

The version of CircuitPython I tested on is: `Adafruit CircuitPython 8.1.0-beta.0-2-g965caa030 on 2023-03-01; SparkFun Pro nRF52840 Mini with nRF52840` and the Bluefruit LE Connect version is `v3.3.2`

RetiredWizard, 2023-03-03:
It appears that this issue only occurs if the UART module is the first module selected, i.e. if you choose the Control Pad module first and then go to the UART module you don't see the problem.

antonio-openroad, 2023-03-14:
I'm trying to reproduce the bug but I always get the message first time UART module is selected. 
I attached a video to verify that we are doing the same test. 

https://user-images.githubusercontent.com/3625420/225076034-09f494c4-a1bb-4c38-9fd8-49d71b3f293f.mp4

Board: Adafruit CircuitPython 8.0.0-beta.0-5-g65ec12afd on 2022-08-21; Adafruit Feather nRF52840 Express with nRF52840
Libraries: adafruit-circuitpython-bundle-8-2.x-mpy-20230314
Android app version: v3.3.2
Phones used : Pixel 3a Android 12, Pixel 6 Android 13

Could you confirm that the steps used to reproduce the bug in the video are correct?

RetiredWizard, 2023-03-14:
It looks like the steps you are performing are correct. 

I'm using an older Moto phone running Android 9 and figured that was going to turn out to be the issue but I just tried using an ItsyBitsy nrf52840 and like you didn't see the problem so I guess it's specific to the Sparkfun Pro nRF52840 which is sort of surprising.

I'll do some more testing to see if any other clues show up.

RetiredWizard, 2023-03-14:
Well it really looks to me like it's specific to the Sparkfun. One interesting thing I noticed is that on the Sparkfun when it's not responding (the first entry to the UART module) the normal print statement to the console works
```python
        uart_service.write(("Echoing: ->"+cmd+"<- Enter Q to quit").encode('utf-8'))
        uart_service.write(b'\n')
        print(cmd)
```
even though the two `uart_service.write` don't display anything to the App, so the `uart_service.read` is picking up the message from the App.

RetiredWizard, 2023-03-15:
I threw the SparkFun UF2 on the ItsyBitsy and didn't see the problem. The ItsyBitsy UF2 didn't boot up the FLASH on the SparkFun board but I did find that the Arduino Nano BLE UF2 ran. The problem still occurred. It looks to me like there's something in the SparkFun's hardware which is causing the board not to transmit initially. 

Are there any bluetooth signals sent to the board from the App when one of the App modules is loaded or exited?

I tested each of the modules by simply entering the module, returning to the module list and then entering the UART module and seeing if I received the Echo response from the test script above:

Info: No
UART: Yes
Plotter: Yes
Pin I/O: Crashed test script `UnicodeError:`
Controller: Yes
Thermal Camera: Yes
Image Transfer: No

RetiredWizard, 2023-05-06:
I've tested this again on another nrf52840 board (seeed sense) and don't see the issue. As this appears to be limited to the Sparkfun board and there's an easy work around, I'm thinking it's not worth running down. 

Thanks for taking a look and trying to reproduce this!

closed 2023-05-06 by RetiredWizard