adafruit/Bluefruit_LE_Connect_Android_V2 #24: Bluefruit Connect App not Working with new Android OS

reparke, 2023-03-11:
I use the Bluefruit Connect App in my class, and this semester some of the students who use Android phones are seeing error messages when they try to install it (I've attach two examples). The message looks like the app isn't compatible with newer versions of Android.

Do you know of a workaround, or can the app be updated to be compatible with the Android OS?

Thanks!

ladyada, 2023-03-11:
@antonio-openroad i think you worked on this app? lmk!

reparke, 2023-03-11:
Oops! I forgot to attached the screenshots.

[image]
[image]

antonio-openroad, 2023-03-14:
> @antonio-openroad i think you worked on this app? lmk!
Yes, I developed that app. 

Google Play now needs that apps are updated approximately every 2 years :
> Starting from 1 Nov 2022, apps that don't target an API level within two years of the latest Android release (API 29 and below) won't be available to new users with devices running Android versions newer than your app's target API level. This means that new users won't be able to discover or install your app on Google Play

I am updating the app from API 29 (Android 10) to API 33 (Android 13) which is the latest published Android API version.

PR #25 (2023-03-17): Android12 new-permissions
Updated Android targetSdkVersion from 29 to 33
Updated external libraries including Nordic DFU library
Added support for new Bluetooth permissions introduced in Android 12
Fixes #24
closed 2023-03-17 by antonio-openroad